rpng/open_vins #446: Maybe an error in propogate.cpp when construct discrete noise covariance matrix

WZG3661, 2024-05-08:
In `Propagate.cpp`:
https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/state/Propagator.cpp#L459-L466
why the covariance of bias is divided by `dt` not multiplied
but in the doc of IMU Propagation:
[image]
the covariance of bias is multiplied by `dt`
and in `fast_state_propagate`:
https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/state/Propagator.cpp#L220-L228

a4msaran, 2024-05-09:
The missing dt is added in G in the function _compute_F_and_G_analytic_ and _compute_F_and_G_discrete_

https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/state/Propagator.cpp#L960C2-L961

a4msaran, 2024-05-09:
The missing dt is added in G in the function _compute_F_and_G_analytic_ and _compute_F_and_G_discrete_

https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/state/Propagator.cpp#L960-L961

WZG3661, 2024-05-09:
Thanks a lot!

closed 2024-05-09 by WZG3661