rpng/open_vins #449: EuRoC MAV Dataset Broken rosbag2 Download Links

chasekunz, 2024-06-06:
Hi,

Thank you for aggregating so many datasets in your docs! For the EuRoC MAV Dataset the rosbag2 links are broken. Can those be fixed?

Datasets page: [https://docs.openvins.com/gs-datasets.html](https://docs.openvins.com/gs-datasets.html)

bsutherland333, 2024-09-04:
In the meantime you can download the [ROS1 versions](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) and convert them using the [instructions for converting them to ROS2 bags](https://docs.openvins.com/dev-ros1-to-ros2.html).

goldbattle, 2024-09-08:
Should be addressed in https://github.com/rpng/open_vins/pull/468, apologies for that.

closed 2024-12-15 by goldbattle

joanvallve, 2026-10-19:
It is happening again. Could you please fix the links? Thanks