rpng/open_vins #451: It seems that the code computing chi2 is not correct

NewThinker-Jeffrey, 2024-06-12:
Relevant code:

https://github.com/rpng/open_vins/blame/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/update/UpdaterSLAM.cpp#L395

Now it's as below:

```
    double chi2 = res.dot(S.llt().solve(res));
```

I think it should be like:
```
    double transformed_res = S.llt().solve(res);
    double chi2 = transformed_res.dot(transformed_res);
```

Other similar code:

https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/update/UpdaterMSCKF.cpp#L212

https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/update/UpdaterZeroVelocity.cpp#L206

goldbattle, 2024-11-23:
Utlimently we want to solve:

$$ res^\top * S^{-1} * res $$

So we can first solve for the last half of the equation

$$ res^\top * (S^{-1} * res) = res^\top * x$$

$$ x= S^{-1} * res $$
$$  S * x = res $$

This last equation can be solved via backsolving
https://eigen.tuxfamily.org/dox/group__TutorialLinearAlgebra.html

```
Eigen::Matrix3f A;
Eigen::Vector3f b;
Eigen::Vector3f x = A.colPivHouseholderQr().solve(b);
```

In this case we are using the LLT since we "should" have symmetric PSD measurement innovation.
[image]

NewThinker-Jeffrey, 2024-11-24:
Sorry, it was my mistake. I incorrectly interpreted `S.llt().solve(res)` as `S.llt().matrixL().solve(res)`.

closed 2024-11-24 by NewThinker-Jeffrey