rpng/open_vins #453: failed static init

Jiangjiacheng-scau, 2024-06-18:
[init]: failed static init: no accel jerk detected, platform moving too much

how to solve it?

precyon, 2024-06-19:
On what datastet? Or are you using your own?

If you are using your own, what does the start of the dataset look like? The error seems to suggest that the setup is already moving at the start? 

You could:
1. Post the full log for the community to look at and better understand what's going on. You could set the verbosity to `DEBUG`.
2. Look at the disparities and IMU values/jerk at the start tweak the `init_*`  options in `estimator_config.yaml`

goldbattle, 2024-08-12:
hi @Jiangjiacheng-scau , here are some steps to get your static initialization tuned. The values will likely be specific to your IMU and robotic platform's vibrations.

- Start the device start at static and set the config `init_imu_thresh` to a very large number
- You can then take a look at the debug messages to see what the current value is
- Pickup the device and look at how the numbers have changed
- Set the `init_imu_thresh` to a value which is a bit larger then the static amount (prevent false positives)

closed 2024-11-23 by goldbattle