rpng/open_vins #456: KLT:false and only tracking with ORB-descriptors? (OpenVins not working for ORB-descriptor alone)

f5ces27er5c767s, 2024-07-10:
Hi there @goldbattle thank you so much for making this code accessible for everyone, it is great using it to study more on visual-inertial systems.

I am messaging about the config mode where KLT: false and only ORB-descriptor feature tracking is done. I am getting this error when turning the KLT tracking off and only depending on feature tracking between frames:

```
[DEBUG] Final Keypoints Size: 60
[DEBUG] Final Descriptors Size: 60x32
[DEBUG] Final Descriptors Type: 0
[DEBUG] Number of new keypoints: 60
[DEBUG] New descriptors size: 60x32
[DEBUG] New descriptors type: 0
[DEBUG] Converting descriptors to CV_32F
[DEBUG] Converted desc_last size: 57x32
[DEBUG] Converted desc_new size: 60x32
[DEBUG] Converted desc_last type: 5
[DEBUG] Converted desc_new type: 5
[ERROR] Descriptor matrices have different numbers of rows. Cannot match keypoints.
[DEBUG] Resized descriptors to have the same number of rows: 57
[DEBUG] Matching keypoints temporally
OpenCV Error: Assertion failed ((type == CV_8U && dtype == CV_32S) || dtype == CV_32F) in batchDistance, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/stat.cpp, line 3941
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/stat.cpp:3941: error: (-215) (type == CV_8U && dtype == CV_32S) || dtype == CV_32F in function batchDistance

================================================================================REQUIRED process [ov_msckf-1] has died!
process has died [pid 25595, exit code -6, cmd /home/user/catkin_ws/devel/lib/ov_msckf/run_subscribe_msckf __name:=ov_msckf __log:=/home/user/.ros/log/7b90806e-3e30-11ef-bdbe-0013eff20e63/ov_msckf-1.log].
log file: /home/user/.ros/log/7b90806e-3e30-11ef-bdbe-0013eff20e63/ov_msckf-1*.log
Initiating shutdown!
================================================================================
[ov_msckf-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
user@user-Legion-Pro-5-16IRX8:~/catkin_ws$
```

Please ignore the debug print statements above, I was just trying to see what the ORB-descriptor sizes were and making sure to match their size. The original code seems not to work with feature descriptor-alone (with KLT: false), do you have any suggestions about how to fix this, or what code is needed?

Thank you so much, I appreciate it @goldbattle please let me know when you have some time to answer :)

goldbattle, 2024-07-10:
Seems pretty weird. Do you have any example data that this happens on?

f5ces27er5c767s, 2024-07-21:
Hi @goldbattle thank you for your quick reply and apologies for the late response :) I resolved this issue, I will close this.

Opening new thread, thanks~

closed 2024-07-21 by f5ces27er5c767s

omardiaa, 2025-02-21:
@f5ces27er5c767s What did you do, I think I am facing the same issue here? Once I disable klt and rebuild using `colcon build`, I get 

```
[run_subscribe_msckf-1]   what():  OpenCV(4.5.4) ./modules/core/src/batch_distance.cpp:274: error: (-215:Assertion failed) type == src2.type() && src1.cols == src2.cols && (type == CV_32F || type == CV_8U) in function 'batchDistance'
[run_subscribe_msckf-1]
[ERROR] [run_subscribe_msckf-1]: process has died [pid 21275, exit code -6, cmd '/catkin_ws/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ov_msckf --params-file /tmp/launch_params_hn0p4ws4 --params-file /tmp/launch_params_6e7zsmb7 --params-file /tmp/launch_params_48jfn8qs --params-file /tmp/launch_params_czgp7yh1 --params-file /tmp/launch_params_ccqbc4qn'].
[run_subscribe_msckf-1] [TIME]: 0.0064 seconds total (156.
```

Yanhof, 2025-09-29:
I face the same issue, @omardiaa and @f5ces27er5c767s did you find a solution?