rpng/open_vins #458: Can someone help in connecting openvins with px4 for publishing the data?

ShreyasKuntnal, 2024-07-22:
I am a bit new to this. Checked out px4 documentation but was filling to find the right file to do changes in. So can someone help please?

goldbattle, 2024-07-27:
I know that when we tried in the past there is some issues with getting good timestamps from PX4 IMU stream. You will likely need to figure out how to get an IMU timestamp that is in the same clock as whatever camera you are using.

ShreyasKuntnal, 2024-07-29:
@goldbattle Thank you so much for the reply and keeping this repository active.
Hey px4 is bit now far site, Can you please help in reducing the drift in d435i Camera please?

[image]
This is using ZUPT set to true. I am just rotating the camera in same place but still there is lot of drift. 
If i set ZUPT false, This is what it looks like:
[image]

For your kind information I have calibration cam and imu using kalibr and alliance ros. 
These are my configuration files:
estimator_config.yaml:
```
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: false # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 1 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: false # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 11 # how many clones in the sliding window
max_slam: 50 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 40 # how many MSCKF features to use in the update
dt_slam_delay: 1 # delay before initializing (helps with stability from bad initialization...)

gravity_mag: 9.81 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: false
zupt_chi2_multipler: 0.6 # set to 0 for only disp-based
zupt_max_velocity: 0.5
zupt_noise_multiplier: 10
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 1.0 # how many seconds to collect initialization information
init_imu_thresh: 2.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

[... 2152 lines cut ...]
VioManager.cpp:620 [TIME]: 0.0017 seconds for re-tri & marg (11 clones in state)
VioManager.cpp:628 [TIME]: 0.006005 seconds for total (camera 0)
VioManager.cpp:654 q_GtoI = -0.370,0.406,-0.497,0.672 | p_IinG = -258.793,-224.739,-57.132 | dist = 347.83 (meters)
VioManager.cpp:657 bg = -0.0273,0.0002,-0.0131 | ba = -0.0086,0.0704,0.0089
VioManager.cpp:662 camera-imu timeoffset = 0.01640
VioManager.cpp:669 cam0 intrinsics = 601.606,600.750,331.221,247.343 | 0.093,-0.209,-0.001,0.001
VioManager.cpp:678 cam0 extrinsics = 0.002,0.003,0.003,1.000 | 0.019,-0.004,-0.009
ROS1Visualizer.cpp:483 [TIME]: 0.0066 seconds total (152.5 hz, 1.66 ms behind)
ROS1Visualizer.cpp:356 camera-imu timeoffset = 0.01640

ROS1Visualizer.cpp:363 cam0 intrinsics:
ROS1Visualizer.cpp:364 601.606,600.750,331.221,247.343
ROS1Visualizer.cpp:365 0.09326,-0.20936,-0.00071,0.00136

ROS1Visualizer.cpp:376 T_C0toI:
ROS1Visualizer.cpp:377 1.000,-0.007,0.007,-0.019,
ROS1Visualizer.cpp:378 0.007,1.000,-0.005,0.004,
ROS1Visualizer.cpp:379 -0.007,0.005,1.000,0.009,
ROS1Visualizer.cpp:380 0.000,0.000,0.000,1.000

ROS1Visualizer.cpp:435 TIME: 18.510 seconds
```

You can find my calibration files here:
[Calibration Files Folder](https://drive.google.com/drive/folders/1hR_MxBLU5vu8E_1GEvaO-zFruonA93Nr?usp=sharing)

goldbattle, 2025-11-30:
Some details in these threads
https://discuss.px4.io/t/publishing-vio-data-to-fmu-in-vehicle-visual-odometry-in-ros-2-python/45536
https://docs.px4.io/main/en/ros2/offboard_control

If someone wants to post a complete example, that would be cool if they can get it to work.