rpng/open_vins #467: on jetson xavier, drift happens with ros2 but not with ros

omeredemen, 2024-09-04:
I have a jetson xavier which have jetpack 4.6.2 built on it. My sistem includes:
- ros2 galactic (from source)
- realsense (debian packages)
- realsense_ros (from source)
- openvins

I have a realsense d456 and i start the camera with following command
`ros2 run realsense2_camera realsense2_camera_node \
--ros-args -r __ns:=/d456
-r __node:=d456
-p unite_imu_method:=2
-p enable_infra1:=true
-p enable_infra2:=true
-p depth_module.infra_profile:=640x480x30
-p depth_module.depth_profile:=640x480x30
-p pointcloud.enable:=true
-p enable_gyro:=true
-p enable_accel:=true
-p gyro_fps:=200
-p accel_fps:=200
-p depth_module.laser_power:=150.0
-p enable_color:=false`

and i start ov_msckf with 
`ros2 launch ov_msckf subscribe.launch.py config_path:=/home/xavier/rs_d456_stereo/estimator_config.yaml
`
here is my estimator_config.yaml and the other kalibr file that i am using for configuration: 
[rs_d456_stereo.zip](https://github.com/user-attachments/files/16862058/rs_d456_stereo.zip)

Odometry is very well at first but it always drifts after some time (5-10 min or less). Even if i don't make a move after initialization. But i am not facing this issue with ros1 (i tried with noetic and melodic). I also use openvins with ROS2 Humble on my pc without any problem.

I examined ROS1Visualizer.cpp and ROS2Visualizer.cpp but i didn't see logical difference between ros1 and ros2.

Is this issue has been faced by other users? I couldn't solve this.

mo-aqeel, 2024-09-26:
Hi, did you face any problem while building the package in jetson?

omeredemen, 2024-09-26:
well i could build without problem except i should have been increase my swap file because jetson needs extra memory to build openvins.