rpng/open_vins #473: Openvins build error in jetson orin nano

mo-aqeel, 2024-09-26:
Hello, I am trying to build OpenVINS on my Jetson Orin Nano. During the build process, I encountered some errors related to the ArUco module. To resolve this, I disabled ArUco support while running the build. Now I am facing errors related cv::mat.

System Information:

    Operating System: Ubuntu 20.04
    ROS Version: Noetic
    OpenCV Version: 4.2 (ROS)

**Error** 

catkin build -DENABLE_ARUCO_TAGS=OFF
-------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/nvidia/catkin_ws/devel:/opt/ros/noetic
Workspace:                   /home/nvidia/workspace
-------------------------------------------------------------------------
Build Space:        [exists] /home/nvidia/workspace/build
Devel Space:        [exists] /home/nvidia/workspace/devel
Install Space:      [unused] /home/nvidia/workspace/install
Log Space:          [exists] /home/nvidia/workspace/logs
Source Space:       [exists] /home/nvidia/workspace/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------
Additional CMake Args:       -DENABLE_ARUCO_TAGS=OFF
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------
[build] Found 5 packages in 0.0 seconds.                                                                                                                                                        
[build] Package table is up to date.                                                                                                                                                            
Starting  >>> ov_core                                                                                                                                                                           
Starting  >>> ov_data                                                                                                                                                                           
Finished  <<< ov_data                 [ 1.7 seconds ]                                                                                                                                           
________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << ov_core:cmake /home/nvidia/workspace/logs/ov_core/build.cmake.001.log                                                                                                             
CMake Warning at /home/nvidia/workspace/src/open_vins/ov_core/CMakeLists.txt:18 (message):
  DISABLING ARUCOTAG TRACKING!

CMake Warning at /home/nvidia/workspace/src/open_vins/ov_core/cmake/ROS1.cmake:89 (add_executable):
  Cannot generate a safe runtime search path for target test_webcam because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libopencv_calib3d.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_dnn.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_features2d.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_flann.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_highgui.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_ml.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_objdetect.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
    runtime library [libopencv_photo.so.4.2] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /usr/local/lib
[... 162 lines cut ...]
make[2]: *** [CMakeFiles/test_tracking.dir/build.make:263: /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_tracking] Error 1
make[1]: *** [CMakeFiles/Makefile2:1234: CMakeFiles/test_tracking.dir/all] Error 2
/usr/bin/ld: CMakeFiles/test_profile.dir/src/test_profile.cpp.o: in function `main':
/home/nvidia/workspace/src/open_vins/ov_core/src/test_profile.cpp:92: undefined reference to `cv::Mat::Mat(int, int, int)'
/usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_profile.cpp:93: undefined reference to `cv::Mat::Mat(int, int, int)'
/usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::Mat::Mat(int, int, int, cv::Scalar_<double> const&)'
/usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::Mat::Mat()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_profile.dir/build.make:263: /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_profile] Error 1
make[1]: *** [CMakeFiles/Makefile2:1072: CMakeFiles/test_profile.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/nvidia/workspace/build/ov_core; catkin build --get-env ov_core | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

................................................................................................................................................................................................
Failed     << ov_core:make            [ Exited with code 2 ]                                                                                                                                    
Failed    <<< ov_core                 [ 4 minutes and 17.5 seconds ]                                                                                                                            
Abandoned <<< ov_eval                 [ Unrelated job failed ]                                                                                                                                  
Abandoned <<< ov_init                 [ Unrelated job failed ]                                                                                                                                  
Abandoned <<< ov_msckf                [ Unrelated job failed ]                                                                                                                                  
[build] Summary: 1 of 5 packages succeeded.                                                                                                                                                     
[build]   Ignored:   None.                                                                                                                                                                      
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                        
[build]   Abandoned: 3 packages were abandoned.                                                                                                                                                 
[build]   Failed:    1 packages failed.                                                                                                                                                         
[build] Runtime: 4 minutes and 17.8 seconds total.

zs1013, 2024-11-01:
> 您好，我正在尝试在我的 Jetson Orin Nano 上构建 OpenVINS。在构建过程中，我遇到了一些与 ArUco 模块相关的错误。为了解决这个问题，我在运行构建时禁用了 ArUco 支持。现在我遇到了与 cv：：mat 相关的错误。
> 
>  系统信息：
> 
> ```
> Operating System: Ubuntu 20.04
> ROS Version: Noetic
> OpenCV Version: 4.2 (ROS)
> ```
> 
> ** 错误**
> 
> catkin build -DENABLE_ARUCO_TAGS=关
> 
> 配置文件：默认扩展：[cached] /home/nvidia/catkin_ws/devel：/opt/ros/noetic工作区：/home/nvidia/workspace
> 
> 构建空间： [exists] /home/nvidia/workspace/build开发空间： [存在] /home/nvidia/workspace/devel安装空间：[未使用] /home/nvidia/workspace/install日志空间： [exists] /home/nvidia/workspace/logs源空间： [exists] /home/nvidia/workspace/srcDESTDIR： [未使用] 无开发空间布局：链接安装空间布局：无其他 CMake 参数：-DENABLE_ARUCO_TAGS=OFF其他 Make 参数：无其他 catkin Make 参数：无内部 Make Job 服务器：True缓存作业环境：False-------------------------------------------------------------------------
> 
> Buildlisted 包：无已跳过的软件包：无
> 
> 工作区配置显示有效。注意：强制 CMake 为每个包运行。[构建]在 0.0 秒内找到 5 个包裹。[构建]Package 表是最新的。正在启动 >>> ov_core正在启动 >>> ov_data完成 <<< ov_data [ 1.7 秒 ]
> 
> 警告 << ov_core：cmake /home/nvidia/workspace/logs/ov_core/build.cmake.001.log/home/nvidia/workspace/src/open_vins/ov_core/CMakeLists.txt：18 的 CMake 警告（消息）：禁用 ARUCOTAG 跟踪！
> 
> cd /home/nvidia/workspace/build/ov_core;catkin build --get-env ov_core |catkin env -si /usr/bin/cmake /home/nvidia/workspace/src/open_vins/ov_core --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/nvidia/workspace/devel/.private/ov_core -DCMAKE_INSTALL_PREFIX=/home/nvidia/workspace/install -DENABLE_ARUCO_TAGS=OFF;光盘 -
> 
> ................................................................................................................................................................................................
> 
> 错误 << ov_core：make /home/nvidia/workspace/logs/ov_core/build.make.002.log/usr/bin/ld： CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o： 在函数 `ov_core::CamRadtan::undistort_f(Eigen::Matrix<float, 2, 1, 0, 2, 1> const&)': /home/nvidia/workspace/src/open_vins/ov_core/src/cam/CamRadtan.h:106: undefined reference to ` cv：：Mat：：Mat（int， int， int）' 中/usr/bin/ld： CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o： 在函数 `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp:175: undefined reference to ` cv：：Mat：：Mat（）' 中/usr/bin/ld： CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o： 在函数 `cv::MatExpr::operator cv::Mat() const': /usr/include/opencv4/opencv2/core/mat.inl.hpp:3077: undefined reference to ` cv：：Mat：：Mat（）' 中/usr/bin/ld： CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o： 在函数 `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp:200: undefined reference to ` cv：：Mat：：Mat（）' 中/usr/bin/ld： /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp：200： 对 `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to ` cv：：Mat：：Mat（int， int， int， cv：：Scalar_ const&）' 的未定义引用collect2：错误：ld 返回 1 退出状态make[2]： *** [CMakeFiles/test_webcam.dir/build.make：263： /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_webcam] 错误 1make[1]：*** [CMakeFiles/Makefile2：181：CMakeFiles/test_webcam.dir/all] 错误 2make[1]： *** 正在等待未完成的工作..../usr/bin/ld： CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o： 在函数 `cv::MatExpr::operator cv::Mat() const': /usr/include/opencv4/opencv2/core/mat.inl.hpp:3077: undefined reference to ` cv：：Mat：：Mat（）' 中/usr/bin/ld： CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o： 在函数 `handle_stereo(double, double, cv::Mat, cv::Mat)': /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:318: undefined reference to ` cv：：Mat：：Mat（）' 中/usr/bin/ld： /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp：318： 对 `cv::Mat::Mat()' /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function ` ov_core：：CamRadtan：：undistort_f（Eigen：：Matrix const&）' 的未定义引用：/home/nvidia/workspace/src/open_vins/ov_core/src/cam/CamRadtan.h：106：对 main 的 `cv::Mat::Mat(int, int, int)' /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function ` 未定义引用：/home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp：220：对 `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:220: undefined reference to ` cv：：Mat：：Mat（） 的未定义引用'/usr/bin/ld： /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so： 对 main 的 `cv::Mat::Mat(int, int, int, cv::Scalar_<double> const&)' collect2: error: ld returned 1 exit status make[2]: *** [CMakeFiles/test_tracking.dir/build.make:263: /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_tracking] Error 1 make[1]: *** [CMakeFiles/Makefile2:1234: CMakeFiles/test_tracking.dir/all] Error 2 /usr/bin/ld: CMakeFiles/test_profile.dir/src/test_profile.cpp.o: in function ` 未定义引用'：/home/nvidia/workspace/src/open_vins/ov_core/src/test_profile.cpp：92：对 `cv::Mat::Mat(int, int, int)' /usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_profile.cpp:93: undefined reference to ` cv：：Mat：：Mat（int， int， int）' 的未定义引用/usr/bin/ld： /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so： 对 `cv::Mat::Mat(int, int, int, cv::Scalar_<double> const&)' /usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to ` cv：：Mat：：Mat（） 的未定义引用'collect2：错误：ld 返回 1 退出状态make[2]： *** [CMakeFiles/test_profile.dir/build.make：263： /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_profile] 错误 1make[1]：*** [CMakeFiles/Makefile2：1072：CMakeFiles/test_profile.dir/all] 错误 2make： *** [Makefile：141： all] 错误 2cd /home/n [... cut ...]
> ................................................................................................................................................................................................失败的 << ov_core：make [ 退出，代码 2 ]失败 <<< ov_core [ 4 分 17.5 秒 ]已放弃 <<< ov_eval [ 无关作业失败 ]已放弃 <<< ov_init [ 无关作业失败 ]已放弃 <<< ov_msckf [ 无关作业失败 ][构建]摘要：5 个程序包中的 1 个成功。[构建] 忽略：无。[构建] 警告：1 个程序包成功，但出现警告。[构建] Abandoned：放弃了 3 个包。[构建] 失败：1 个包失败。[构建]运行时间：总共 4 分 17.8 秒。

May I ask if you solved it? What was your method?

zs1013, 2024-11-01:
> Hello, I am trying to build OpenVINS on my Jetson Orin Nano. During the build process, I encountered some errors related to the ArUco module. To resolve this, I disabled ArUco support while running the build. Now I am facing errors related cv::mat.
> 
> System Information:
> 
> ```
> Operating System: Ubuntu 20.04
> ROS Version: Noetic
> OpenCV Version: 4.2 (ROS)
> ```
> 
> **Error**
> 
> catkin build -DENABLE_ARUCO_TAGS=OFF
> 
> Profile: default Extending: [cached] /home/nvidia/catkin_ws/devel:/opt/ros/noetic Workspace: /home/nvidia/workspace
> 
> Build Space: [exists] /home/nvidia/workspace/build Devel Space: [exists] /home/nvidia/workspace/devel Install Space: [unused] /home/nvidia/workspace/install Log Space: [exists] /home/nvidia/workspace/logs Source Space: [exists] /home/nvidia/workspace/src DESTDIR: [unused] None Devel Space Layout: linked Install Space Layout: None Additional CMake Args: -DENABLE_ARUCO_TAGS=OFF Additional Make Args: None Additional catkin Make Args: None Internal Make Job Server: True Cache Job Environments: False-------------------------------------------------------------------------
> 
> Buildlisted Packages: None Skiplisted Packages: None
> 
> Workspace configuration appears valid. NOTE: Forcing CMake to run for each package. [build] Found 5 packages in 0.0 seconds. [build] Package table is up to date. Starting >>> ov_core Starting >>> ov_data Finished <<< ov_data [ 1.7 seconds ]
> 
> Warnings << ov_core:cmake /home/nvidia/workspace/logs/ov_core/build.cmake.001.log CMake Warning at /home/nvidia/workspace/src/open_vins/ov_core/CMakeLists.txt:18 (message): DISABLING ARUCOTAG TRACKING!
> 
> cd /home/nvidia/workspace/build/ov_core; catkin build --get-env ov_core | catkin env -si /usr/bin/cmake /home/nvidia/workspace/src/open_vins/ov_core --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/nvidia/workspace/devel/.private/ov_core -DCMAKE_INSTALL_PREFIX=/home/nvidia/workspace/install -DENABLE_ARUCO_TAGS=OFF; cd -
> 
> ................................................................................................................................................................................................
> 
> Errors << ov_core:make /home/nvidia/workspace/logs/ov_core/build.make.002.log /usr/bin/ld: CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o: in function `ov_core::CamRadtan::undistort_f(Eigen::Matrix<float, 2, 1, 0, 2, 1> const&)': /home/nvidia/workspace/src/open_vins/ov_core/src/cam/CamRadtan.h:106: undefined reference to `cv::Mat::Mat(int, int, int)' /usr/bin/ld: CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o: in function `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp:175: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o: in function `cv::MatExpr::operator cv::Mat() const': /usr/include/opencv4/opencv2/core/mat.inl.hpp:3077: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: CMakeFiles/test_webcam.dir/src/test_webcam.cpp.o: in function `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp:200: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_webcam.cpp:200: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::Mat::Mat(int, int, int, cv::Scalar_ const&)' collect2: error: ld returned 1 exit status make[2]: *** [CMakeFiles/test_webcam.dir/build.make:263: /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_webcam] Error 1 make[1]: *** [CMakeFiles/Makefile2:181: CMakeFiles/test_webcam.dir/all] Error 2 make[1]: *** Waiting for unfinished jobs.... /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function `cv::MatExpr::operator cv::Mat() const': /usr/include/opencv4/opencv2/core/mat.inl.hpp:3077: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function `handle_stereo(double, double, cv::Mat, cv::Mat)': /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:318: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:318: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function `ov_core::CamRadtan::undistort_f(Eigen::Matrix<float, 2, 1, 0, 2, 1> const&)': /home/nvidia/workspace/src/open_vins/ov_core/src/cam/CamRadtan.h:106: undefined reference to `cv::Mat::Mat(int, int, int)' /usr/bin/ld: CMakeFiles/test_tracking.dir/src/test_tracking.cpp.o: in function `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:220: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/src/open_vins/ov_core/src/test_tracking.cpp:220: undefined reference to `cv::Mat::Mat()' /usr/bin/ld: /home/nvidia/workspace/devel/.private/ov_core/lib/libov_core_lib.so: undefined reference to `cv::Mat::Mat(int, int, int, cv::Scalar_<double> const&)' collect2: error: ld returned 1 exit status make[2]: *** [CMakeFiles/test_tracking.dir/build.make:263: /home/nvidia/workspace/devel/.private/ov_core/lib/ov_core/test_tracking] Error 1 make[1]: *** [CMakeFiles/Makefile2:1234: CMakeFiles/test_tracking.dir/all] Error 2 /usr/bin/ld: CMakeFiles/test_profile.dir/src/test_profile.cpp.o: in function `main': /home/nvidia/workspace/src/open_vins/ov_core/src/test_profile.cpp:92: undefined reference to `cv::M [... cut ...]  2 ] Failed <<< ov_core [ 4 minutes and 17.5 seconds ] Abandoned <<< ov_eval [ Unrelated job failed ] Abandoned <<< ov_init [ Unrelated job failed ] Abandoned <<< ov_msckf [ Unrelated job failed ] [build] Summary: 1 of 5 packages succeeded. [build] Ignored: None. [build] Warnings: 1 packages succeeded with warnings. [build] Abandoned: 3 packages were abandoned. [build] Failed: 1 packages failed. [build] Runtime: 4 minutes and 17.8 seconds total.

May I ask if you solved it? What was your method?

mo-aqeel, 2024-11-02:
Hello, If possible, try re-flashing the Jetson and starting everything from
scratch. Worked for me.