rpng/open_vins #475: Odom Drift

VivekMangeUD, 2024-09-30:
Hello
I am trying to run Openvins on my system and after initialization the system starts to drift.

After Initializing, With ZUPT as true, While the system is stationary, everything works fine, But when the system is moving there is a huge drift on ODOM path. 

I have attached the config file, Kalibr files, and screen recording of the test on the google drive - https://drive.google.com/drive/folders/1hH7HIAgSKzxHoH2Tnc82iHsJx88-7CEq?usp=sharing

I see that the drift after initialization is a common issue faced by a lot of user, Can anyone help out with what should i change in the system.

goldbattle, 2024-11-23:
What sensors are you using to record this data? Some custom sensors or?

goldbattle, 2025-11-30:
Yes those are reasonable reprojections, but the IMU frequency likely means there is still some issue with the IMU driver that should be fixed.