rpng/open_vins #476: D455 initial test

camron-BHF, 2024-10-02:
Hi there, I'm trying to use OpenVINS for VIO with my realsense d455 camera. I was looking through the issues/tutorial on youtube and I'm able to get it 'working' except when I move the camera slightly, it just keeps on going in that direction.

I used this command: `roslaunch ov_msckf subscribe.launch config:=rs_d455 max_cameras:=1` 

And this was the output I received:

[init]: successful initialization in 0.0008 seconds
[init]: orientation = -0.4500, 0.3623, -0.6072, 0.5454
[init]: bias gyro = -0.0710, 0.4919, -0.0374
[init]: velocity = 0.0000, 0.0000, 0.0000
[init]: bias accel = 1.4682, -9.0378, 3.2274
[init]: position = 0.0000, 0.0000, 0.0000
[TIME]: 0.0251 seconds total (39.8 hz, 29.20 ms behind)
[TIME]: 0.0136 seconds total (73.6 hz, 204.21 ms behind)
[TIME]: 0.0184 seconds total (54.5 hz, 199.29 ms behind)
[TIME]: 0.0227 seconds total (44.1 hz, 195.95 ms behind)
q_GtoI = -0.659,0.191,-0.467,0.558 | p_IinG = 0.032,0.613,-0.012 | dist = 0.00 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0342 seconds total (29.3 hz, 195.28 ms behind)
q_GtoI = -0.609,-0.310,0.313,0.660 | p_IinG = -1.293,1.668,1.439 | dist = 2.23 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0841 seconds total (11.9 hz, 8.36 ms behind)
q_GtoI = -0.592,-0.321,0.332,0.661 | p_IinG = -1.293,1.675,1.485 | dist = 2.28 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0372 seconds total (26.9 hz, 17.03 ms behind)
q_GtoI = -0.577,-0.329,0.354,0.659 | p_IinG = -1.290,1.684,1.531 | dist = 2.32 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0286 seconds total (35.0 hz, 13.69 ms behind)
q_GtoI = -0.563,-0.339,0.370,0.657 | p_IinG = -1.286,1.695,1.577 | dist = 2.37 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0304 seconds total (32.9 hz, 17.36 ms behind)
q_GtoI = -0.549,-0.352,0.387,0.652 | p_IinG = -1.280,1.708,1.624 | dist = 2.42 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0217 seconds total (46.0 hz, 14.02 ms behind)
q_GtoI = -0.533,-0.369,0.405,0.645 | p_IinG = -1.272,1.724,1.670 | dist = 2.47 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0233 seconds total (42.9 hz, 10.69 ms behind)
q_GtoI = -0.517,-0.386,0.421,0.637 | p_IinG = -1.262,1.741,1.716 | dist = 2.52 (meters)
bg = -0.0710,0.4919,-0.0374 | ba = 1.4682,-9.0378,3.2274
camera-imu timeoffset = 0.00252
cam0 intrinsics = 416.852,414.921,421.025,237.762 | -0.046,0.034,-0.000,0.000
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.028,-0.003,-0.024
[TIME]: 0.0228 seconds total (43.8 hz, 15.60 ms behind)
q_GtoI = -0.505,-0.370,0.472,0.621 | p_IinG = 1.575,2.932,1.843 | dist = 2.64 (meters)
bg = -0.0381,0.5023,-0.0423 | ba = 1.4673,-9.0385,3.2257
camera-imu timeoffset = -0.00984
cam0 intrinsics = 416.841,414.917,421.063,237.767 | -0.046,0.034,-0.000,-0.001
cam0 extrinsics = -0.000,0.002,0.004,1.000 | -0.027,-0.003,-0.023
[TIME]: 0.0208 seconds total (48.2 hz, 15.52 ms behind)
q_GtoI = -0.550,-0.348,0.448,0.613 | p_IinG = 1.771,4.756,1.929 | dist = 2.95 (meters)
bg = -0.0109,0.4718,-0.0145 | ba = 1.4683,-9.0380,3.2247
camera-imu timeoffset = -0.00828
cam0 intrinsics = 416.558,414.919,421.032,237.839 | -0.040,0.044,0.003,0.013
cam0 extrinsics = -0.001,0.002,0.004,1.000 | -0.028,-0.003,-0.023
[TIME]: 0.0238 seconds total (42.0 hz, 17.42 ms behind)
q_GtoI = -0.535,-0.365,0.461,0.606 | p_IinG = 1.857,4.871,1.977 | dist = 3.10 (meters)
bg = -0.0109,0.4718,-0.0145 | ba = 1.4683,-9.0380,3.2247
camera-imu timeoffset = -0.00828
cam0 intrinsics = 416.558,414.919,421.032,237.839 | -0.040,0.044,0.003,0.013
cam0 extrinsics = -0.001,0.002,0.004,1.000 | -0.028,-0.003,-0.023
[TIME]: 0.0166 seconds total (60.3 hz, 20.43 ms behind)
q_GtoI = -0.523,-0.378,0.472,0.601 | p_IinG = 1.942,4.986,2.022 | dist = 3.25 (meters)
bg = -0.0109,0.4718,-0.0145 | ba = 1.4683,-9.0380,3.2247
camera-imu timeoffset = -0.00828
cam0 intrinsics = 416.558,414.919,421.032,237.839 | -0.040,0.044,0.003,0.013
cam0 extrinsics = -0.001,0.002,0.004,1.000 | -0.028,-0.003,-0.023
[TIME]: 0.0151 seconds total (66.4 hz, 25.60 ms behind)
q_GtoI = -0.511,-0.390,0.482,0.596 | p_IinG = 2.029,5.105,2.066 | dist = 3.40 (meters)
bg = -0.0109,0.4718,-0.0145 | ba = 1.4683,-9.0380,3.2247
camera-imu timeoffset = -0.00828
cam0 intrinsics = 416.558,414.919,421.032,237.839 | -0.040,0.044,0.003,0.013
cam0 extrinsics = -0.001,0.002,0.004,1.000 | -0.028,-0.003,-0.023
[TIME]: 0.0140 seconds total (71.3 hz, 27.76 ms behind)
q_GtoI = -0.499,-0.402,0.491,0.590 | p_IinG = 2.117,5.228,2.110 | dist = 3.56 (meters)
bg = -0.0109,0.4718,-0.0145 | ba = 1.4683,-9.0380,3.2247
camera-imu timeoffset = -0.00828
cam0 intrinsics = 416.558,414.919,421.032,237.839 | -0.040,0.044,0.003,0.013
cam0 extrinsics = -0.001,0.002,0.004,1.000 | -0.028,-0.003,-0.023
[TIME]: 0.0197 seconds total (50.7 hz, 41.94 ms behind)
q_GtoI = -0.453,-0.445,0.537,0.556 | p_IinG = 2.237,5.620,2.073 | dist = 4.07 (meters)
bg = -0.0056,0.4905,-0.0173 | ba = 1.4730,-9.0576,3.2270
camera-imu timeoffset = -0.00739
cam0 intrinsics = 416.244,414.863,421.070,237.711 | -0.039,0.047,0.004,0.012
cam0 extrinsics = -0.001,0.001,0.003,1.000 | -0.029,-0.003,-0.024
[TIME]: 0.0145 seconds total (68.9 hz, 33.92 ms behind)
q_GtoI = -0.421,-0.475,0.556,0.537 | p_IinG = 2.581,6.039,2.215 | dist = 4.63 (meters)
bg = -0.0075,0.4872,-0.0192 | ba = 1.4721,-9.0520,3.2262
camera-imu timeoffset = -0.00733
cam0 intrinsics = 416.226,414.878,421.079,237.786 | -0.039,0.047,0.004,0.012
cam0 extrinsics = -0.000,0.001,0.003,1.000 | -0.029,-0.003,-0.024
[TIME]: 0.0213 seconds total (46.9 hz, 24.82 ms behind)
q_GtoI = -0.398,-0.495,0.567,0.524 | p_IinG = 2.794,6.372,2.283 | dist = 5.03 (meters)
bg = -0.0075,0.4872,-0.0192 | ba = 1.4721,-9.0520,3.2262
camera-imu timeoffset = -0.00733
cam0 intrinsics = 416.226,414.878,421.079,237.786 | -0.039,0.047,0.004,0.012
cam0 extrinsics = -0.000,0.001,0.003,1.000 | -0.029,-0.003,-0.024
[TIME]: 0.0159 seconds total (62.8 hz, 18.15 ms behind)
q_GtoI = -0.350,-0.536,0.588,0.494 | p_IinG = 3.243,7.114,2.406 | dist = 5.90 (meters)
bg = -0.0075,0.4872,-0.0192 | ba = 1.4721,-9.0520,3.2262
camera-imu timeoffset = -0.00733
cam0 intrinsics = 416.226,414.878,421.079,237.786 | -0.039,0.047,0.004,0.012
cam0 extrinsics = -0.000,0.001,0.003,1.000 | -0.029,-0.003,-0.024
[TIME]: 0.0228 seconds total (43.8 hz, 10.54 ms behind)
q_GtoI = -0.340,-0.546,0.592,0.485 | p_IinG = 3.359,7.317,2.435 | dist = 6.14 (meters)
bg = -0.0075,0.4872,-0.0192 | ba = 1.4721,-9.0520,3.2262
camera-imu timeoffset = -0.00733
cam0 intrinsics = 416.226,414.878,421.079,237.786 | -0.039,0.047,0.004,0.012
cam0 extrinsics = -0.000,0.001,0.003,1.000 | -0.029,-0.003,-0.024

[image]

Even though all I did was rotate the camera in my hand.

I am working on Ubuntu 18.04 with ROS Melodic, running it in my own environment not a docker container

TiagoMartinss, 2024-10-15:
I'm also getting a similar issue [here](https://github.com/rpng/open_vins/issues/478), i can apply translations with my camera but once i apply a rotation it instantly stop working well, i also did setup and calibrate my camera and imu on a melodic 18.04 ROS Docker.

goldbattle, 2024-11-23:
Are you using the config without modification from the repository? I should work out of the box unless the sensor had some big change since its release (e.g. IMU moved or something).

tmrpj, 2024-12-05:
I have the same issue using the default config. if I bump the camera into the desk it will begin to drift continuously in one direction. I can stop it with zupt, but that's not wanted since the robot its supposed to be attached to has regular bumps.

MJavadZallaghi, 2025-04-30:
The same issue for me with D455. The sensor flew hundreds of meters when I kept it in my hand. Also, most of the time it gets stuck in the init phase.

TiagoMartinss, 2025-04-30:
I'm using ROS 2 Humble with a RealSense D455, running the default OpenVINS configuration in mono mode.

Like others, I also encountered the issue where the sensor "flies" hundreds of meters in Rviz even when held still, and the system often gets stuck in the initialization phase.

Here’s what helped me:

- I enabled ZUPT to avoid the system being stuck during init.
- I made sure that both the .launch.py and the config file are set consistently to mono mode.
- To initialize properly, I hold the D455 in my hand and wait for the init phase to finish.

Once initialized, I slowly rotate the camera sideways (left or right). This usually allows the SLAM points (visible as green in the image view or red in the 3D Rviz viewer) to appear and stabilize. 

One issue I still have:

I don't see the MSCKF points (orange) as shown in the OpenVINS tutorial video. They flash very briefly in Rviz but are not maintained as in the demo.

Additionally:

To get proper motion tracking, I found that I must translate and rotate the camera at the same time—pure rotation or pure translation alone tends to break tracking.

Hope this helps others facing similar issues.

imedpsd, 2025-05-26:
Hello guys I didn't had much of a trouble when I re did the calibration with the rs_d455 Im facing a heck of a problem in simulation but the camera it selfs works very fine 

[image]

omardiaa, 2025-06-26:
Anyone could fix the issue? I am having the same issue. It is 2025 no response!