rpng/open_vins #477: Issue saving the simulation state with save_total_state in ROS2

harry-graves, 2024-10-04:
Hi, I noticed an issue when trying to save the simulation state in a ROS2 environment. After changing `filepath_est`, `filepath_std` and `filepath_gt`, when using the `save_total_state:=true` flag, I get the following error:

```
[run_subscribe_msckf-1] terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
[run_subscribe_msckf-1]   what():  boost::filesystem::create_directories: Invalid argument
[ERROR] [run_subscribe_msckf-1]: process has died [pid 611689, exit code -6, cmd '/home/ori/ros2_ws/install/ov_msckf/lib/ov_msckf/run_subscribe_msckf --ros-args -r __ns:=/ov_msckf --params-file /tmp/launch_params_f1udo1j0 --params-file /tmp/launch_params_8cgjp7ug --params-file /tmp/launch_params__47f7d76 --params-file /tmp/launch_params_vj12ovz1 --params-file /tmp/launch_params_7y3x4w5v'].
```

However I noticed this can be worked around by manually replacing `"state_estimate.txt"`, `"state_deviation.txt"` and `"state_groundtruth.txt"` with your own full filepaths in the following file:

https://github.com/rpng/open_vins/blob/17b73cfe4b870ade0a65f9eb217d8aab58deae19/ov_msckf/src/ros/ROS2Visualizer.cpp#L108-L110

This would then give me a stderr output when rebuilding, but it worked as a temporary fix.

diplodocuslongus, 2024-11-08:
> After changing filepath_est, filepath_std and filepath_gt

Where do you change these files? 
I have the same error as yours when setting `save_total_state:=true` (ROS2 humble) but without changing any destination file path or file name.

Your fix does the trick to save the states though, thanks a lot.

harry-graves, 2024-11-08:
`filepath_est`, `filepath_std` and `filepath_gt` can be changed in `estimator_config.yaml`, though whatever filepaths you put here are overwritten by the fix I have given above.

I'm not too sure what you mean by saving the groundtruth, but the above workaround allowed me to save the estimated state, which includes the trajectory, to a text file.

diplodocuslongus, 2024-11-12:
> not too sure what you mean by saving the groundtruth

I was referring to the ground truth data that is included in a dataset when running OpenVINS offline. 
And I meant that when running openVINS with a pre-existing dataset (e.g. EuROC), I don't see any `state_groundtruth.txt` file.

Also the data format saved in the estimated state file is different from that expected by ov_eval for, e.g., trajectory alignment.

goldbattle, 2024-11-23:
I think this needs to be fixed. I think the ROS2 version of this saving doesn't work as expected.