rpng/open_vins #481: Running OpenVINS for very long durations

tianyilim, 2024-11-02:
Hello, thanks for the good work and great code!

I've been trying to get OpenVINS to run on large-scale, long datasets, in particular the [Vision Benchmark in Rome](https://github.com/rvp-group/vbr-devkit) dataset. This dataset is quite interesting as it contains handheld sequences > 20min and > 2km in length.

Initially, OpenVINS seems to do quite well, but as the trajectory gets longer, I find the covariance estimates also increase, and I guess without place recognition techniques to "anchor" it, the trajectory eventually drifts away [Youtube video](https://youtu.be/TVcD7GX1vio).

Maybe this is a limitation of a pure EKF-based VIO system, but are there any things that i can try to tackle this issue? E.g force reinitialization when covariance increases, tuning configs, etc.? Thanks in advance!

<details>

<summary>Overall Config</summary>

```yaml
%YAML:1.0 # need to specify the file type at the top!

verbosity: "INFO" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: true # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 2 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: true # if the transform between camera and IMU should be optimized R_ItoC, p_CinI
calib_cam_intrinsics: true # if camera intrinsics should be optimized (focal, center, distortion)
calib_cam_timeoffset: true # if timeoffset between camera and IMU should be optimized
calib_imu_intrinsics: true # if imu intrinsics should be calibrated (rotation and skew-scale matrix)
calib_imu_g_sensitivity: false # if gyroscope gravity sensitivity (Tg) should be calibrated

max_clones: 20 # how many clones in the sliding window
# ! set a bit larger hopefully for local 'loop closure'
max_slam: 100 # number of features in our state vector
max_slam_in_update: 25 # update can be split into sequential updates of batches, how many in a batch
max_msckf_in_update: 50 # how many MSCKF features to use in the update
# ! VBR Spagna initializes okay
dt_slam_delay: 0.0 # delay before initializing (helps with stability from bad initialization...)
# ! VBR Spagna weirdness in grav vector
gravity_mag: 9.7 # magnitude of gravity in this location

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "ANCHORED_MSCKF_INVERSE_DEPTH"
feat_rep_aruco: "ANCHORED_MSCKF_INVERSE_DEPTH"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: false
zupt_chi2_multipler: 0 # set to 0 for only disp-based
zupt_max_velocity: 0.05
zupt_noise_multiplier: 50
zupt_max_disparity: 0.5 # set to 0 for only imu-based
zupt_only_at_beginning: true

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 1.5 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 50 # how many features to track during initialization (saves on computation)

init_dyn_use: true # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 8 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
[... 145 lines cut ...]
  cam_overlaps: [1]
  camera_model: pinhole
  distortion_coeffs: [-0.15442153959026667, 0.22415792664801637, 0.00028611532170761544, -0.0011553024948397013]
  distortion_model: radtan
  intrinsics: [1265.4157426547047, 1259.125033829179, 649.1764873455104, 349.11875004082356]
  resolution: [1388, 700]
  rostopic: /camera_left/image_raw
  timeshift_cam_imu: 0.0014439866415333613
cam1:
  T_imu_cam:
    - [ 0.047533765306, 0.002997724519, 0.998865133441, 0.117518620394]
    - [ -0.998851236543, -0.005926227938, 0.047550889366, -0.235758955147]
    - [ 0.006062046927, -0.999977946492, 0.002712584915, -0.066699484195]
    - [ 0.000000000000, 0.000000000000, 0.000000000000, 1.000000000000]
  cam_overlaps: [0]
  camera_model: pinhole
  distortion_coeffs: [-0.13964919973164264, 0.3032246444678759, -0.0006841993380782645, -0.0017521213083201822]
  distortion_model: radtan
  intrinsics: [1270.980680475146, 1266.6167801504557, 692.2080529147544, 346.53773474480096]
  resolution: [1388, 700]
  rostopic: /camera_right/image_raw
  timeshift_cam_imu: 0.0015213183603639074
```

</details>

tianyilim, 2024-11-03:
As an initial strategy, I wanted to reset the filter state, and use the last-known state as a initialization point.

Looking at #398, re-starting the ROS node is not an option for me.

I modified the Serial ROS1 bag reader code to attempt this, I'm not too sure if I did this properly -- but if someone could sanity check my impl, it would be greatly appreciated :)

https://github.com/tianyilim/open_vins

omeredemen, 2024-11-11:
well this is a good job. actually i wanted to do something like that.

omeredemen, 2024-11-11:
it is a good work thanks for sharing. i wanted to do a reset state like you did. but i haven't started yet.
can you share what was the result you get with your updates?

goldbattle, 2024-11-23:
This might be the covariance is becoming ill-defined, or the statistical chi2 checks start to fail at that point. Could you take a look at that point and see if the filter stops getting feature updates (MSCKF and SLAM)? 

I think one concern with your approach is that you will remove all the tracked features, so basically have non-continuous pose tracking. Probably a better method is to "update" with a fake GPS / mocap sensors to reduce the uncertainty of the system to remove this instability (if you want to avoid this issue with the global uncertainty being unbounded).

goldbattle, 2025-11-30:
Feel free to checkout https://github.com/rpng/sqrtVINS which should be able to run much longer out of the box.