rpng/open_vins #496: Error:(-49:Unknown error code -49) Input file is invalid in function 'open' when i try to run ov_msckf launch file.

aviralshastri, 2025-03-15:
Okay, so i built my workspace with OpenVINS for usage in my project but when i try launching ov_msckf it shows this error:

```shell
aviral@aviral-LOQ-15IRX9:~$ roslaunch ov_msckf subscribe.launch config_path:=/home/aviral/rs_custom/estimator_config.yaml
... logging to /home/aviral/.ros/log/e178b504-01d5-11f0-b22f-e5ec57e4c527/roslaunch-aviral-LOQ-15IRX9-23648.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://aviral-LOQ-15IRX9:33487/

SUMMARY
========

CLEAR PARAMETERS
 * /ov_msckf/

PARAMETERS
 * /ov_msckf/config_path: /home/aviral/rs_c...
 * /ov_msckf/max_cameras: 2
 * /ov_msckf/record_timing_filepath: /tmp/traj_timing.txt
 * /ov_msckf/record_timing_information: False
 * /ov_msckf/use_stereo: True
 * /ov_msckf/verbosity: INFO
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    ov_msckf (ov_msckf/run_subscribe_msckf)

ROS_MASTER_URI=http://localhost:11311

process[ov_msckf-1]: started with pid [23663]
overriding node verbosity with value from ROS!
Setting printing level to: INFO
overriding node max_cameras with value from ROS!
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/persistence.cpp:719: error: (-49:Unknown error code -49) Input file is invalid in function 'open'

================================================================================REQUIRED process [ov_msckf-1] has died!
process has died [pid 23663, exit code -6, cmd /home/aviral/catkin_ws/devel/lib/ov_msckf/run_subscribe_msckf __name:=ov_msckf __log:=/home/aviral/.ros/log/e178b504-01d5-11f0-b22f-e5ec57e4c527/ov_msckf-1.log].
log file: /home/aviral/.ros/log/e178b504-01d5-11f0-b22f-e5ec57e4c527/ov_msckf-1*.log
Initiating shutdown!
================================================================================
[ov_msckf-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
aviral@aviral-LOQ-15IRX9:~$ 

``` 

[image]

### This is my ov_msckf launch command:
```shell
roslaunch ov_msckf subscribe.launch config_path:=/home/aviral/rs_custom/estimator_config.yaml
```
### This is my rviz launch command:
```shell
rviz -d catkin_ws/src/open_vins/ov_msckf/launch/display.rviz
```

### This is my ros2 realsense launch command with args:

```shell
ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_gyro:=true -p enable_accel:=true -p gyro_fps:=400 -p accel_fps:=200 -p align_depth.enable:=true -p depth_width:=848 -p depth_height:=480 -p enable_depth:=true -p align_depth:=true -p ublish_odom_tf:=true -p unite_imu_method:=linear_interpolation
```

[... 124 lines cut ...]
downsize_aruco: true

# ==================================================================
# ==================================================================

# camera noises and chi-squared threshold multipliers
up_msckf_sigma_px: 1
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

# masks for our images
use_mask: false

# imu and camera spacial-temporal
# imu config should also have the correct noise values
relative_config_imu: "kalibr_imu_chain.yaml"
relative_config_imucam: "kalibr_imucam_chain.yaml"
``` 

This is my first time opening an issue on github so if i missed something to add here that can be helpful in determining plz tell and i will update it asap.

Quick response would be great as i am implementing vio in my project and the deadline is very close :)

JingMohan, 2025-03-26:
I have similar problem.

goldbattle, 2025-03-29:
Make sure you have `%YAML:1.0 # need to specify the file type at the top!` at the top of each yaml file.

j11b0, 2025-08-28:
I have the same problem too. I have `%YAML:1.0` on each config but no help.

goldbattle, 2025-11-29:
Hmm, I am not sure, can the process access the file to be able to read it? This yaml from opencv is typically the most abstruse.