rpng/open_vins #498: ㅇ

edfafde2, 2025-03-28:
[image]

[image]

[image]

(base) imlab105@imlab105-TRX40-AORUS-PRO-WIFI:~/25v2x/dong_openvins$ ros2 topic info 

/ov_msckf/pathgt
Type: nav_msgs/msg/Path
Publisher count: 1
Subscription count: 1

Hello, I sincerely appreciate you creating this algorithm. I am trying to use it, but I have encountered the following issues:

1. I executed the MH_01_easy tutorial, but the path_gt is not visualized in RViz as shown in the image. However, I checked, and both the subscriber and publisher are present. What could be the reason it is not displaying?

2. Is pathimu the estimated position of the system through the OpenVINS algorithm?

goldbattle, 2025-11-29:
This might be an issue with the ros2 launch python file. I think I have only tested it with the ros1 version. Were you able to get it working?

goldbattle, 2025-11-29:
See https://github.com/rpng/open_vins/issues/502

closed 2025-11-29 by goldbattle