rpng/open_vins #505: real time / online usage

Mehmetmrtgl, 2025-04-30:
hello, how can i use openvins online with my robot's zed1 camera and xsens imu, odometry data. I want to run open vins with the data and visualize it when the robot moves.

lectronuser, 2025-05-29:
I understood that your pose always drifts. Right?
If it is It can be related with your imu frequency because it doesn't seem stable. The imu frequency should be constant as possible as it can be. I can see the imu period in your pdf file as 10ms but with a 20ms standart deviation value which is too high.
A new calibration with higher and straight imu frequency (200Hz) might help you

omeredemen, 2025-05-29:
I understood that your pose always drifts. Right?
If it is It can be related with your imu frequency because it doesn't seem stable. The imu frequency should be constant as possible as it can be. I can see the imu period in your pdf file as 10ms but with a 20ms standart deviation value which is too high.
A new calibration with higher and straight imu frequency (200Hz) might help you

goldbattle, 2025-11-29:
Is ZED2 a rolling shutter or global shutter? If you move at different speeds do you get some results? I.e. at the start you should be able to see pure integration. If you don't see the camera frame rotating then maybe some issue. Are you able to record a rosbag and replay it using the serial version?

goldbattle, 2025-11-30:
There is this old issue also about ZED2 maybe some issue with this platform?
https://github.com/rpng/open_vins/issues/450

Mehmetmrtgl, 2025-11-30:
Thank you for your response. When I used Zed2i's own IMU, I was able to run OpenVins properly. I think I couldn't run it with the Xsens IMU because I couldn't calibrate it properly with an external IMU due to Zed2i's rolling shutter. I don't know if this is a correct or incorrect assumption.

goldbattle, 2025-11-30:
Great good to know. What driver were you using with the ZED2i?