rpng/open_vins #507: how to save

xym0601, 2025-05-13:
```
record_timing_information: true # if we want to record timing information of the method
record_timing_filepath: "/catkin_ws/src/open_vins/tmp/traj_timing.txt" # https://docs.openvins.com/eval-timing.html#eval-ov-timing-flame

# if we want to save the simulation state and its diagional covariance
# use this with rosrun ov_eval error_simulation
save_total_state: true
filepath_est: "/catkin_ws/src/open_vins/tmp/ov_estimate.txt"
filepath_std: "/catkin_ws/src/open_vins/tmp/ov_estimate_std.txt"
filepath_gt: "/catkin_ws/src/open_vins/tmp/ov_groundtruth.txt"
```
I set up autosave, but it doesn't save when I rosplay a bag.why?

goldbattle, 2025-11-29:
The launch files typically overwrite these parameters, so you need to set it there also to enable or comment out those parts.

goldbattle, 2025-11-30:
If you are using ROS2 see: https://github.com/rpng/open_vins/issues/477

DanielChaseButterfield, 2026-01-07:
At least for ROS2, the parameters in the configuration files state saving aren't used at all, as they aren't turned into ROS2 parameters and ROS2Visualizer.cpp won't see them.

I just commented them out of my config.