rpng/open_vins #513: Custom dataset at 100 meters above ground does not perform well

Rubbishwang, 2025-06-21:
Hello, I'm using a self-collected UAV flight dataset at an altitude of around 100m, with the IMU frame defined as Right-Front-Up (RFU). After successful initialization, the navigation solution diverges very quickly. Below are some of my parameters. Could this be due to inappropriate parameter settings, or is the code not well-suited for operations at this altitude?
```yaml
%YAML:1.0 # need to specify the file type at the top!

verbosity: "ALL" # ALL, DEBUG, INFO, WARNING, ERROR, SILENT

use_fej: true # if first-estimate Jacobians should be used (enable for good consistency)
integration: "rk4" # discrete, rk4, analytical (if rk4 or analytical used then analytical covariance propagation is used)
use_stereo: false # if we have more than 1 camera, if we should try to track stereo constraints between pairs
max_cameras: 1 # how many cameras we have 1 = mono, 2 = stereo, >2 = binocular (all mono tracking)

calib_cam_extrinsics: false
calib_cam_intrinsics: false
calib_cam_timeoffset: false
calib_imu_intrinsics: false
calib_imu_g_sensitivity: false

max_clones: 11
max_slam: 50
max_slam_in_update: 25
max_msckf_in_update: 40
dt_slam_delay: 2

gravity_mag: 9.794 # kalibr calibration

feat_rep_msckf: "GLOBAL_3D"
feat_rep_slam: "GLOBAL_3D"
feat_rep_aruco: "GLOBAL_3D"

# zero velocity update parameters we can use
# we support either IMU-based or disparity detection.
try_zupt: false
zupt_chi2_multipler: 1 # set to 0 for only disp-based
zupt_max_velocity: 0.1
zupt_noise_multiplier: 20
zupt_max_disparity: 0 # set to 0 for only imu-based
zupt_only_at_beginning: false

# ==================================================================
# ==================================================================

init_window_time: 2.0 # how many seconds to collect initialization information
init_imu_thresh: 3.3 # threshold for variance of the accelerometer to detect a "jerk" in motion
init_max_disparity: 10.0 # max disparity to consider the platform stationary (dependent on resolution)
init_max_features: 75 # how many features to track during initialization (saves on computation)

init_dyn_use: false # if dynamic initialization should be used
init_dyn_mle_opt_calib: false # if we should optimize calibration during intialization (not recommended)
init_dyn_mle_max_iter: 50 # how many iterations the MLE refinement should use (zero to skip the MLE)
init_dyn_mle_max_time: 0.05 # how many seconds the MLE should be completed in
init_dyn_mle_max_threads: 6 # how many threads the MLE should use
init_dyn_num_pose: 6 # number of poses to use within our window time (evenly spaced)
init_dyn_min_deg: 10.0 # orientation change needed to try to init

init_dyn_inflation_ori: 10 # what to inflate the recovered q_GtoI covariance by
init_dyn_inflation_vel: 100 # what to inflate the recovered v_IinG covariance by
init_dyn_inflation_bg: 10 # what to inflate the recovered bias_g covariance by
init_dyn_inflation_ba: 100 # what to inflate the recovered bias_a covariance by
init_dyn_min_rec_cond: 1e-12 # reciprocal condition number thresh for info inversion

init_dyn_bias_g: [ 0.0017452, 0.0017452, 0.0017452 ] # initial gyroscope bias guess 0.1 deg
init_dyn_bias_a: [ 0.000294, 0.000294, 0.000294 ] # initial accelerometer bias guess 30ug

# ==================================================================
# ==================================================================

record_timing_information: false
record_timing_filepath:""

save_total_state: true
filepath_est:""
filepath_std:""
filepath_gt: ""

# ==================================================================
# ==================================================================

# our front-end feature tracking parameters
# we have a KLT and descriptor based (KLT is better implemented...)
use_klt: true # if true we will use KLT, otherwise use a ORB descriptor + robust matching
num_pts: 200 # number of points (per camera) we will extract and try to track
fast_threshold: 20 # threshold for fast extraction (warning: lower threshs can be expensive)
grid_x: 5 # extraction sub-grid count for horizontal direction (uniform tracking)
grid_y: 5 # extraction sub-grid count for vertical direction (uniform tracking)
min_px_dist: 10 # distance between features (features near each other provide less information)
knn_ratio: 0.8 # descriptor knn threshold for the top two descriptor matches
track_frequency: 50.0 # frequency we will perform feature tracking at (in frames per second / hertz)
downsample_cameras: false # will downsample image in half if true
num_opencv_threads: 4 # -1: auto, 0-1: serial, >1: number of threads
histogram_method: "HISTOGRAM" # NONE, HISTOGRAM, CLAHE

# aruco tag tracker for the system
# DICT_6X6_1000 from https://chev.me/arucogen/
use_aruco: false
num_aruco: 1024
downsize_aruco: true

# ==================================================================
# ==================================================================

# camera noises and chi-squared threshold multipliers
up_msckf_sigma_px: 1.5
up_msckf_chi2_multipler: 1
up_slam_sigma_px: 1.5
up_slam_chi2_multipler: 1
up_aruco_sigma_px: 1
up_aruco_chi2_multipler: 1

# masks for our images
use_mask: false

# imu and camera spacial-temporal
# imu config should also have the correct noise values
relative_config_imu: "kalibr_imu_chain.yaml"
relative_config_imucam: "kalibr_imucam_chain.yaml"
```

[image]

Looking forward to your answer.

hasantahabagci, 2025-06-29:
Did you try on lower altitude like 50-60? I am now encountering same problem, OV works well on lower altitude but it is not working on higher.

hasantahabagci, 2025-07-02:
Check that issue [https://github.com/rpng/open_vins/issues/515](url)

AnielAlexa, 2025-09-07:
Hi, did you managed to solve the issue? I'm curious about using it for same application. Thank you, Aniel

BearrGod, 2025-10-21:
Hello. Check feature initializer options in ov_core. it seams thre is a hidden parameter fi_max_dist or something like that .

goldbattle, 2025-11-30:
How does the feature tracking look to you? Are you able to share a dataset + config you are running with? Seems like a pretty new style of dataset so would be interesting to see what issues are there. Are features being triangulated and being included in the update?